wadpac/GGIR #1253: Implement functionality to read Matrix device files (format: .bin)

jhmigueles, 2025-01-22:
**Is your feature request related to a problem? Please describe.**
Matrix devices store binary (bin) files, currently reading of these files is not supported by GGIR or GGIRread.

**Describe the solution you'd like**
Implementation of reading of bin files stored by Matrix devices.

**Describe alternatives you've considered**
The manufacturer has a python script to convert bin files into csv files. These csv files can be read in GGIR through the `read.myacc.csv` functionality, but it is cumbersome to convert files to csv.

vincentvanhees, 2025-01-22:
Hi Jairo, one general observation. If possible, could you try to refer to this device in the code with a specific and recognisable name. The term "matrix" already has a meaning in R context, which would make the code difficult to read if we also use Matrix in this context. Also when searching online for a sensor named Matrix it is impossible to find anything based on that name alone.

How about including the name of the manufacturer and abbreviating the word matrix? For example, if the sensor is made by company Abc then maybe refer to it as AbcMTX, AbcMat? In the documentation you could then clarify in detail what it is.

vincentvanhees, 2025-01-23:
Ok, I will wait for the PR in the GGIRread repo first. In that way I will also be able to understand a later PR for GGIR.

Maybe also rename `readMatrix` to `readParmayMatrix`? It is an internal function, so for GGIR users it will not make a difference. I am only concerned about readability for GGIR(read) code contributors.

Btw - I noticed you tried to rebuild the gh-pages in your branch. I think it is easier for both of us if we do not do this in branches, instead I think it is best if only I do it in the master branch prior to each new release. In this way we never risk merge problems with the hundreds of files related to the gh-pages.

jhmigueles, 2025-01-23:
> Maybe also rename readMatrix to readParmayMatrix?

Yes, makes sense, done.

> Btw - I noticed you tried to rebuild the gh-pages in your branch.

That's right, should I go back to the previous version? Not sure if that's possible since I already pushed the changes, sorry about that.

vincentvanhees, 2025-01-23:
> should I go back to the previous version?
In GitHub desktop you can undo a specific commits, I think that is the easier solution. Just make sure the PR does not have change to all those html files in it.

PR #1286 (2025-04-04): Issue1253 read matrix functions

Fixes #1253 => Implements functionality to read and process Parmay Matrix sensors data.

### Checklist before merging:

- [x] Existing tests still work (check by running the test suite, e.g. from RStudio).
- [x] Added tests (if you added functionality) or fixed existing test (if you fixed a bug).
- [x] Clean code has been attempted, e.g. intuitive object names and no code redundancy.
- [x] Documentation updated:
  - [x] Function documentation
  - [x] Chapter vignettes for GitHub IO
  - [x] Vignettes for CRAN
- [x] Corresponding issue tagged in PR message. If no [... cut]
closed 2025-04-04 by vincentvanhees