wadpac/GGIR #1258: Need for more specific terminology when referring to visual reports

vincentvanhees, 2025-01-24:
We have been referring to the GGIR pdf reports as visual report, QC report or time series report. This has been confusing as all of them are visual reports, all of them serve a purpose for quality checking, and all of them show time series. I would like to explore whether we can revise how we name and refer to the reports to be more distinct.

Currently we have:

No. | File name                          | Purpose           |
---- |------------------------------------ | ---------------- | 
1     | results/QC/plots_to_check_data_quality_1.pdf | Basic check for recording length and compliance to study protocol in part 2 |
2     | results/visualisation_sleep.pdf | Basic check of sleep detection and sleep diary in part 4 |
3     | results/file summary reports/Time_report_.....pdf | Elaborate check of derived time series of both data and its classification after part 5 |
4     | results/file summary reports/Report_.....pdf | Old visual report that is not ideal for data quality checking but has been used  |

I am ignoring the 'sleep.qc' pdfs as I think those can eventually be deprecated as they now overlap with report 3.

### Idea A

No. | Proposed new filename       | Proposed way of referring to the file in documentation and in GitHub issues
---- | ------------------------------ | ---------------------------------------------------------------------------------------
1     | part2_report_1.pdf  | Part 2 visual report
2     | part4_report.pdf | Part 4 visual report
3     | Report_v2_....pdf | File summary report v1
4     | Report_v1_....pdf | File summary report v2

**Advantages:**
- It is clear in which GGIR part they were generated
- Name does not put restrictions on future content.

**Disadvantage:**
- Filenames themselves do not say much about what the report represents. 
- Giving version numbers to the reports is not ideal as it gives the impression that any update will result in a new number.

### Idea B

No. | Proposed new filename       | Proposed way of referring to the file in documentation and in GitHub issues
---- | ------------------------------ | ---------------------------------------------------------------------------------------
1     | basic_report_1.pdf  | Basic visual report
2     | sleep_report.pdf | Visual sleep report
3     | general_report....pdf | General visual report
4     | Report_....pdf (same as how it was) | Older visual report

**Advantages:**
- Filenames provide a hint about their contents

**Disadvantage:**
- It is not clear in which GGIR part they are generated

@jhmigueles or anyone else seeing this: Do you have a preference or do you have alternative suggestions?

I may want to implement these updates before the 3.2-0 release.

vincentvanhees, 2025-01-24:
Once we have decided on new naming:

- [ ] Update file names
- [ ] Update documentation references to these filenames
- [ ] Update function names and if applicable unit-tests.

jhmigueles, 2025-01-28:
@forget999 thanks for posting this. In future occassions, please post your specific issues, including a reproducible example, at the [GGIR's google group](https://groups.google.com/g/RpackageGGIR) or create a new specific issue as this is not related to the topic of this issue. 

Regarding the error you are getting, GGIR is struggling to identify what combination of thresholds you would like to see in your visual report. I am now working on a fix for this. In the meantime, you can:

1. Remove the configuration file that you can find at "output_[...]/config.csv". 
2. Include `part6_threshold_combi = "40_100_400"` (adapt this to the combination of thresholds you want to use in the visual report). 
3. Run GGIR again, you should get the visual report now.

jhmigueles, 2025-01-28:
@vincentvanhees 

What about combining the options you provide? Something like...

| No. | Proposed new filename | Proposed way of referring to the file in documentation |
|--------------|-----------------------------|-----------------------------|
| 1 | part2_basic_report.pdf | Part 2 basic report |
| 2 | part4_sleep_report.pdf | Part 4 sleep report |
| 3 | Report\_[...].pdf | General visual report (either older or updated) |

Not sure if we need to differentiate older and newer in the general report name, it can complicate future versions (if any). There are other functionalities that are updated over time without receiving new names, although I can see that it can be useful to log what version is used in the file name... Not sure what is the best option here.

vincentvanhees, 2025-01-28:
@forget999 please do not post questions unrelated to the issue.
If you have new question then just open a new issue.

vincentvanhees, 2025-01-28:
thanks @jhmigueles 

> Not sure if we need to differentiate older and newer in the general report name, it can complicate future versions (if any). There are other functionalities that are updated over time without [...]

We currently generate both the new and old report to avoid a radical change in GGIR and to give user some time to get used to the new report. However, eventually I hope to deprecate the new report, so we just need a temporary solution.

How about including old in the "old" report name, and no old in the new report name.

vincentvanhees, 2025-02-20:
Closing this issue, because:

- In the 3.2-0 release I have now renamed

Report_ => old_report_
Time_report_ => report_

in that way the distinction between old and new is clear, and it makes it easier to eventually deprecate the old report.

- After doing this I realise that for part 2 and 4 pdf reports it may be better to keep names as they have been. The names are maybe not perfect, but they are also not causing major confusion while introducing new files may actually cause confusion.

closed 2025-02-20 by vincentvanhees